Inxton/axsharp #56: [NEW-FEATURE]

PTKu, 2023-02-08:
Having following:

~~~Smalltalk
NAMESPACE MonsterData
    CLASS MonsterBase
        VAR PUBLIC
            ArrayOfBytes : ARRAY[0..100] OF BYTE;
            ArrayOfDrives : ARRAY[0..10] OF DriveBase;
        END_VAR        
    END_CLASS
    CLASS Monster EXTENDS MonsterBase
        VAR PUBLIC
            DriveA : DriveBase;
        END_VAR        
    END_CLASS

    CLASS DriveBase
        VAR PUBLIC
            Position : LREAL;
            Velo : LREAL;
            Acc : LREAL;
            Dcc : LREAL;            
        END_VAR        
    END_CLASS
END_NAMESPACE
~~~

The arrays are now rendered as:

tbc

Specter-13, 2023-02-08:
/cib

PTKu, 2023-02-08:
Branch [56-_NEW-FEATURE_Add_border_when_rendering_arrays_of_complex_types](https://github.com/ix-ax/ix/tree/56-_NEW-FEATURE_Add_border_when_rendering_arrays_of_complex_types) created!

PR #57 (2023-02-08): [NEW-FEATURE] Add border when rendering arrays of complex types
closes #56
closed 2023-02-08 by PTKu