Inxton/axsharp #62: [NEW-FEATURE] Compiler should delete previous generated .cs file, if corresponding plc instance is removed

Specter-13, 2023-02-10:
When I create plc enum structure (or any structure) and its instance for example:

```
TYPE myEnum :
    (
        Unknown,
        Available,
        UnAvailable	
    );
    END_TYPE
```

```
myEnum_instance : myEnum;
```

Compiler will generate `myEnum.cs` file. That is perfectly okay.

But when I completely remove my `myEnum.st` structure from PLC program, C# compiled file `myEnum.cs` is still present and is causing errors.

It would be nice, if compiler would remove unnecessary structures.

PTKu, 2023-02-16:
/cib

PTKu, 2023-02-16:
Branch [62-_NEW-FEATURE_Compiler_should_delete_previous_generated_cs_file_if_corresponding_plc_instance_is_removed](https://github.com/ix-ax/ix/tree/62-_NEW-FEATURE_Compiler_should_delete_previous_generated_cs_file_if_corresponding_plc_instance_is_removed) created!

PR #76 (2023-02-16): [NEW-FEATURE] Compiler should delete previous generated .cs file, if corresponding plc instance is removed
closes #62
closed 2023-02-16 by PTKu