Inxton/axsharp #203: [BUG] Leap year has strange behavior

Brano5, 2023-08-07:
When I try to set a date that is 29.2 or later in a leap year. In PLC it is set correctly, but in Onliner it has different values. As shown in image:

[image]

Specter-13, 2023-09-22:
/cib

Specter-13, 2023-09-25:
# Culprit

This bug occurs, when date value (DATE, LDATE, DATETIME, LDATETIME) from PLC is read over WebApi. 
It happens:

1. when is a leap year and date is after 29th february 
e.g. : 05.20.**2000** ( WebApi will return 05.21.2000 ) 

2. when is a next year after leap year before 28th february
e.g.: 02.13.**2001** (WebApi will return 02.14.2001)

3. 27th february and 28th february a year after leap year read over webapi return same value 

[image]

The culprit is `GetFromBinary(long value)` method in corresponding WebApi primitive files (*WebApiDate.cs*, *WebApiLDate.cs*, *WebApiDatetime.cs*, *WebapiILDatetime.cs*). The returned binary value from WebApi is not correct and these inconsistencies happen.

Reproduced in this branch in `AX_Sharp.cs` test file. 

Plc values are defined in `ax-test-project` in `program.st` file:

[image]

# Temporary fix

Temporary not-fully fix was introduced, where date is checked for a leap year a corresponding corrections are done:
This fix is not working for dates 27th february and 28th february (a year after leap year, eg 2001), as they return same value.

[image]

closed 2025-01-24 by PTKu