Inxton/axsharp #330: [NEW-FEATURE] Poco property of DateTime  has different default value than PLC

kuh0005, 2024-08-20:
The shadow validator validates the time value. By default, this value is set in a C# POCO property to 1.1.0001, which is invalid for the PLC.

PTKu, 2025-01-22:
@kuh0005 do you think it would be sufficient to initialize default of date/time related values in POCO objects to minimal controller value?

kuh0005, 2025-01-23:
I made a test with TIA and PLCSIM. It is possible to write datetime variables with default values that are initialized by C#.

[image]

[image]

When I try to read it from the PLC, an exception is raised.

[image]

[image]

kuh0005, 2025-01-23:
As you can see, some time variables have different values between the PLC and the Blazor view (after batch write).

PTKu, 2025-01-23:
OK. Thanks. We will need to discriminate between target platform (TIA vs AX).

closed 2025-01-23 by PTKu